apps4av/avare #428: Traffic altitude reporting delta should be 2 digits

n251ea, 2021-05-27:
Traffic altitude delta is reported in hundreds of feet, get +10 is +1000, +25 is +2500 ft, but atitudes of under 1000 ft are reported as a single digit eg +9 for +900, this should be reported as +09

reddn, 2021-08-21:
this was a simple fix i believe.. i dont have a build environment or a way to test adsb traffic displays...  but the code is pretty straight forward if you want to test what i did ..

https://github.com/apps4av/avare/compare/master...reddn:patch-1

reddn, 2021-08-21:
also im assuming that you can directly call a string concat with an integer without it throwing a type mismatch exception... why can't python be that easy? (I've never really messed wtih java, but its very C like, so i understand it)