apps4av/avare #433: External Scoped Storage (Storing Chart Data on SD)

danieldjewell, 2021-10-11:
I noticed in a few recent commits (https://github.com/apps4av/avare/commit/f24693d1c5b61534e817d9d63e0713db9f24e6a0 and https://github.com/apps4av/avare/commit/0cc610df5a6e3a88548c89625d64f9e279fd019a) that the storage code was updated to support API 30 ScopedStorage. And that's awesome -- but it was a little concerning that external storage (e.g. SD cards) was completely removed. 

I know that external storage has been a bit of a challenge with the Android API changes but the new APIs didn't remove SD card access - it just made it so that it's per-app by default instead of a free-for-all. (Apps can still request permission to write to random locations on external storage.) 

Given the (somewhat) large amount of data that Avare could potentially use (if you were to download all of the sectionals, high/low IFR enroutes, approach plates, TACs, AIP data, etc.) it sure would be nice to still have the option to use external storage. And from my experience with Avare, I don't think it's a deal-breaker if chart data is stored in an app-specific directory (as I recall, an app-specific directory on internal storage is currently the default). 

Unless I'm mistaken, it appears that choosing external storage with Scoped Storage is actually quite easy? See: https://developer.android.com/training/data-storage/app-specific#external-verify-availability

jguy1987, 2021-12-07:
Any news on the possibility of this being integrated into the app? I love using Avare for Flight Simulation but my tablet doesn't have enough storage without the SD Card to handle all the charts and maps.

reddn, 2021-12-07:
To combat high block value, what about using non-compression zip files
aligned to <128k?  I'm sure there will be some overhead, but something has
to give.

gustafson, 2021-12-07:
@reddn It is my opinion that it would be a bad plan to orient our workflow around something that is limiting (in general) for only a small and shrinking portion of the user base especially when it is a detriment to performance for everybody.  (Slower loading tiles for sure, increased battery usage also, also the cost of complexity).

For better or worse, right or wrong, SD card storage is on the outs and the number of people using it decreases asymptotically toward zero.

There are some workarounds for the technical (i.e., format the card to a better filesystem, root, or other) and for the non-technical (buy a bigger card).  These won't be easy for everybody.  I don't like my answer, but it reflects what I think is the best option we have.

gustafson, 2021-12-07:
An alternative I've thought about in the past is to store tiles as binary blobs in a sqlite3 database. Some of the other EFBs do this (or did do this in the past). I suspect this would have less negative impact than storing in zips and _possibly_ net a small savings in disk space (if block misalignment overhead is larger than sqlite overhead). This would certainly save a lot of disk space for any fat based file system.  But of course this means people wouldn't be able to use any (easy) file managers to mess with their tiles. Also it seems likely that this increases frequency of corruption. It would likely require many sqlite transactions when downloading tiles. Any shutdown during the process might leave the state of the database in questionable shape.  You could have different databases per sectional etc and just overwrite... trading one set of problems for another.