apps4av/avare #481: NMEA USB out

1959c172, 2022-10-11:
Would it be possible to have the autopilot NMEA data sent out via USB along with the Bluetooth out? I see we have adsb USB in already. I've been getting drop outs using Bluetooth and thought it would be more reliable via hardwired. Then we could use a USB to serial converter cable for output.
Let me know,
Thanks

kubark42, 2023-02-20:
I'd love this as well.

1) It reduces complexity of parts required. (BT dongle vs FTDI cable.)
2) It eliminates a source of RF noise. I doubt that BT is causing radio issues in the cockpit, but a BT dongle would ideally be mounted on the rear of the instrument panel, near the autopilot, and in that position it's significantly closer to sensitive analog circuitry.

apps4av, 2023-02-25:
In release 10.2.8.

closed 2023-02-25 by apps4av

kubark42, 2023-02-26:
@apps4av, neat! Is there some documentation on what kind of hardware this was tested on, and what kind of cable we need?

apps4av, 2023-02-26:
Supported devices are here
https://github.com/apps4av/avare/blob/master/app/src/main/res/xml/device_filter.xml

kubark42, 2023-02-26:
@apps4av Perfect.

I'm trying to give it a shot by looking in the IO menu, but I'm not seeing a menu entry which looks like it'd be the right one. All I see which seems relevant is `Auto Pilot via Bluetooth`.

apps4av, 2023-03-01:
You may install from GitHub releases. Or wait for update on the store.
Google is taking very long these days to publish updates.

kubark42, 2023-03-01:
Oh, interesting. I had updated 2 days ago but missed that Google Play wasn't playing ball.

I can now see the `Autopilot via USB` option. Awesome! I'm waiting for the USB OTG module so I can plug my phone into an FTDI cable, but in the meantime I'm guessing that the connection string setting `230400,8,n,1` is:
- the baud rate,
- the data bits,
- the polarity, and 
- the stop bits.

So for an AeroCruze 100 Trutrak autopilot, I would think I should set the string to `9600,8,n,1` to account for the required 9600 baud rate.