apps4av/avare #488: Improvements to Help Page

shosk, 2023-01-17:
Last fall, I worked with Peter G. to make several improvements and corrections to the out-dated Avare Help page. Unfortunately, there was no follow up, so I am posting my draft here. I struck through words to be removed. The yellow text is my proposed change.  The orange text is my comments or suggestions.

I would be happy to work with anyone on this task.

[Help-1.docx](https://github.com/apps4av/avare/files/10439093/Help-1.docx)

apps4av, 2023-01-17:
Thanks for the help. We need someone to proof read for correctness. @John W
SBA ***@***.***> can you help? Thanks

JohnW4SBA, 2023-01-19:
shosk wrote:
 > I would be happy to work with anyone on this task.

Wow, great work!  Thank you for taking this on.  I've set aside some 
time to take a look just now.  Here are a few notes (green BG, comments 
in italics) so far (down to "Main Avare Screens").  Please let me know 
if any of this is helpful and you'd like me to do more when time 
permits.  It's fine as-is, imo - and a welcome overdue update on what we 
had before.

shosk, 2023-01-19:
I don't see an attachment. I barely know anything about github.

shosk, 2023-01-19:
It would probably be useful to incorporate most, if not all, of the New Features into the main section.  This might be another separate project..

JohnW4SBA, 2023-01-21:
Shosk wrote:
 > I don't see an attachment.

Ah. Hadn't noticed that my reply was to github.  Please email me 
directly or in a Forum post and I'll send it again as an email attachment.

 >  I barely know anything about github.

Probably much more than I.  Still, it seems to me best if we collaborate 
directly and then figure out how to update the final onto github.  I've 
edited there before, but it's long enough ago I'd be starting from scratch.

gustafson, 2023-01-21:
> *Attention:* This email is from outside Western Michigan University.
> Use caution when opening links and attachments.

Hey guys, sorry I've not responded if I indicated I was going to help on 
this one.  I lack memory of any details of this.

Shosk, for clarity, the in-app documentation is not something I've 
messed with (well... not in a long time at least).  I restarted the 
online documentation renewal, but for the moment these are separate with 
a vision of future unification.

If you don't know github or git, best is to edit the in-app html file 
directly (Download the source file, keeping it as html. Use a true text 
editor like notepad if you can, as word makes too many changes behind 
the scenes. Don't save as a word document (.docx), rather save as html 
(.html). Then send it to someone who can interact with git.

  * Don't highlight the changes, as doing so prevents automatic merging.
  * Avoid unnecessary changes to carriage returns which make merging
    more difficult.  These can be cleaned up later.
  * Follow these tips and we can easily understand the changes you propose.

Thanks for sticking with it... updating documentation is one of the 
hardest parts of the project!

Pete

shosk, 2023-01-21:
John, I don't know your email address. Please send it to me. I'm at Sam.Hoskins gmail.

Here is a suggestion. Why don't you proof read for content accuracy the original .docx document, posted at the start of this thread? At the same time, I will start working on the HTML source and be making the changes there.  Then we can hand it over to Zubin or Peter when we think it's ready.

JohnW4SBA, 2023-01-21:
> John, I don't know your email address. Please send it to me.

Thanks, Sam.  I've emailed you the docx just now, and look forward to 
working with you via email.  Your work on this is a significant 
contribution to Avare users.

shosk, 2024-09-09:
I did a 400 mile round trip this weekend and gave AvareX a good try. I had
27 loaded for the outbound and 28 for the inbound.

Heading out, early in the flight there was a almost a half hour difference
between the estimated time enroute between my Garmin GPS 196 and AvareX.
The Garmin said it was 2 hours and 15 minutes out and AvareX was calling
over 3 hours. I was carrying another tablet, so I was able to look at the
original Avare,  which agreed with the Garmin.

However , on the way back, there was only about a three minute discrepancy
between the two. So maybe the issue was fixed in version 28.

apps4av, 2024-09-09:
It's not the version.
You must have either old weather, or altitude was selected wrong.

shosk, 2024-11-04:
This may have been discussed before,

Doing downloads from the menu tab,  I frequently get a download failed
notice and have to restart. This morning I had to restart it about eight
times to get the charts updated.

Maybe it has something to do with my internet provider, I have AT&T
wireless with about a 22 mps download speed.

Sam