apps4av/avare #498: kml file are in /data/data/com.ds.avare/files (unreadable by user) and truncated

marcmerlin, 2023-04-18:
```
130|root@deb:/ # find /sdcard | grep -i 'kml$'
1|root@deb:/ # find /data | grep -i 'kml$'                                     
/data/data/com.ds.avare/files/tracks/2022-03-24_17-52-19.KML
/data/data/com.ds.avare/files/tracks/2022-03-24_17-52-45.KML
```
kml files should be under /sdcard/ so that the user can find them.

```
data/com.ds.avare/files/tracks/2023-04*                                       <
-rw------- u0_a88   u0_a88          0 2023-04-03 17:28 2023-04-03_17-28-03.KML
-rw------- u0_a88   u0_a88      90112 2023-04-04 14:54 2023-04-04_10-43-12.KML
-rw------- u0_a88   u0_a88          0 2023-04-09 20:27 2023-04-09_20-27-11.KML
-rw------- u0_a88   u0_a88      98304 2023-04-14 11:01 2023-04-14_08-52-24.KML
-rw------- u0_a88   u0_a88          0 2023-04-14 14:58 2023-04-14_14-58-38.KML
-rw------- u0_a88   u0_a88       8192 2023-04-16 13:03 2023-04-16_11-45-40.KML
-rw------- u0_a88   u0_a88      24576 2023-04-16 19:19 2023-04-16_17-53-03.KML
-rw------- u0_a88   u0_a88          0 2023-04-17 20:16 2023-04-17_20-16-11.KML
```
this shows the files are unreadable by the user unless you have root

```
root@deb:/ # tail /data/data/com.ds.avare/files/tracks/2023-04-16_17-53-03.KML 
					-120.93425750732422,35.75496292114258,2583.000082656
					-120.93949127197266,35.762664794921875,2581.0000825920006
					-120.94397735595703,35.76929473876953,2583.000082656
					-120.9480972290039,35.775390625,2583.000082656
					-120.95217895507813,35.781463623046875,2578.0000824960002
					-120.95736694335938,35.789188385009766,2575.0000824000003
					-120.96187591552734,35.79583740234375,2575.0000824000003
					-120.96599578857422,35.8018913269043,2575.0000824000003
					-120.9700698852539,35.80794143676758,2577.000082464
					-120.97460174560547,35.814506530761root@deb:/ # 
```

Every file I have is either 0 or unterminated

So gpsbabel won't read them:
```
saruman [mc]$ for i in *.kml; do gpsbabel -i kml -f $i -o gpx -F $i.gpx; done
XML Reader:Read error: Premature end of document. (2022-03-25_08-35-05.kml, line 331, col 7)
XML Reader:Read error: Premature end of document. (2022-03-25_13-12-19.kml, line 326, col 10)
XML Reader:Read error: Premature end of document. (2022-03-27_12-04-07.kml, line 330, col 43)
XML Reader:Read error: Premature end of document. (2022-03-27_14-23-58.kml, line 625, col 49)
XML Reader:Read error: Premature end of document. (2022-07-15_12-18-52.kml, line 472, col 7)
XML Reader:Read error: Premature end of document. (2022-08-28_12-41-52.kml, line 768, col 45)
XML Reader:Read error: Premature end of document. (2022-09-05_14-21-10.kml, line 1318, col 35)
XML Reader:Read error: Premature end of document. (2023-04-04_10-43-12.kml, line 1688, col 12)
XML Reader:Read error: Premature end of document. (2023-04-14_08-52-24.kml, line 1833, col 9)
XML Reader:Read error: Premature end of document. (2023-04-16_11-45-40.kml, line 186, col 16)
XML Reader:Read error: Premature end of document. (2023-04-16_17-53-03.kml, line 467, col 40)
```

marcmerlin, 2023-04-18:
the app does not have a version number that I can see when I launch it, but google play says "last updated feb 25 2023"

marcmerlin, 2023-04-18:
So for each kml, I need to remove the corrupted/incomplete data at the end, and then add this
```
    	    </coordinates>
    	</LineString>
        </Placemark>
    </Document>
</kml>
```

After that gpsbabel will be happy as expected, but even then the track has no times in it, so it cannot be used for geotagging or generally knowing where I was at a given time, which i unfortunate.

Is it something you can fix and maybe instead just generate GPX which includes time?

marcmerlin, 2023-10-10:
Ping?

apps4av, 2023-10-11:
Is it not exported via export feature ?