apps4av/avare #506: Cannot enter hexadecimal digits in Acft page: ICAO Mode S code field

SolKatzman, 2024-03-15:
Avare Version 10.3.1
Android Version 14
Samsung Galaxy Tab A7 Lite

In the "Acft" tab, the "ICAO Mode S code" field seems to want a 6 digit number (default value is "123456"), but when you click in the field, a keyboard comes up that only allows for decimal digits. I have tried this with the Samsung keyboard as well as with the Google "Gboard" installed and enabled.

If you go to the aircraft registry, and look up your N-number you will find 2 entries:

Mode S Code (base 8 / Oct) 52332503
Mode S Code (Base 16 / Hex) A9B543

a) Is Avare expecting the Hexadecimal version? Seems most likely.
b) Hex digits are the set {0,1,2,3,4,5,6,7,8,9,A,B,C,D,E,F} How can A-F be entered?
c) What Avare functions make use of this value? It is not a field in an ICAO flight plan.

Thanks for you attention to this.
Sol Katzman

JohnW4SBA, 2024-03-15:
> keyboard comes up that only allows for decimal digits.
So are you having a problem with typing in your base 8 (Octal) aircraft ID?

 > Mode S Code (base 8 / Oct) 52332503 ... Is Avare expecting the 
Hexadecimal version?

Yes, if 52332503 is the S code base 8 number given for your aircraft 
that's the one you want to input for Avare.

In case that's the base 8 S code for your aircraft, I typed that into 
Avare just now.  I noted that the typical way my Android devices work 
for typing differs for that box in Avare, in that I needed to first 
delete the 123456 example pre-set in the box.  I found it easy to do 
that either with 6 taps on the "delete" key; or by tapping on the 123456 
number and then tapping on the "teardrop" shaped blob that hangs down 
from the cursor indicator and then tapping "Select all" and then the 
delete key.

Numeric Base Details:
Everyone's S code number looks different in each base number value, 
often including the number of digits for the same value.  We're 
typically used to base 10 (values 0-9 for each digit), so even the base 
8 value in your email will look quite different in the base 10 system we 
all normally use.  Hexadecimal requires fewer digits than Octal because 
each digit has more possible values.

Minutia:
Base 8 and Hex are derived from how computers work - in "binary" with 
"bit" values of 0 and 1 to correspond to digital switch "on" (1) or 
"off" (0) possible values.  Octal is how computer programmers initially 
abbreviated long base 2 (binary) numbers into "bytes" for machine 
language coding used in the early days.  Hexadecimal also became 
standard for computer work.  Base 10 that we all learned in school arose 
from the number of fingers most people have on both hands.

 > b) Hex digits are the set {0,1,2,3,4,5,6,7,8,9,A,B,C,D,E,F} How can 
A-F be entered?

Apparently it can't.  My guess is that only the numeric keyboard is 
enabled for that box in Avare for simplicity, displaying larger (fewer) 
key buttons, and to reduce the chance of inadvertent input errors.

 > c) What Avare functions make use of this value?

I seldom bother to input that value (I fly various club aircraft) but 
one function know of is preventing your aircraft showing up as a traffic 
conflict in ADS-B.  Because radar is often less accurate than GPS and 
ADS-B by nature is delayed, sometimes ADS-B displays an "aircraft" 
following close near my altitude.

 > Thanks for you attention to this.

I hope something in all this may have been helpful, Sol - and that 
you'll forgive me if it's completely irrelevant. :)

Happy Flying
John

JohnW4SBA, 2024-03-15:
Sol wrote:
 >  >  Is Avare expecting the Hexadecimal version?

  I replied (in error):
 > Yes

I had deleted a line and didn't notice before sending, so it's exactly 
the opposite answer to your question quoted above.  So, if 52332503 is 
the S code base 8 number given for your aircraft that's the one you want 
to input for Avare.

SolKatzman, 2024-03-16:
Dear John,

I will boil down your answers as follows:

1) Avare wants the octal version of the ICAO S code in the Acft page.
2) Avare uses it to suppress the "self" aircraft from ADSB-in traffic.

I will test that when I go to the airport next. There is a thread on the AVS4U google forum inquiring how to suppress the "self" aircraft traffic. This may resolve that.

If it works, I will create a (less urgent) new issue:

Acft page should indicate that numeric octal version should be entered in ICAO S code field.

Incidentally, Issue #383 from Aug 6, 2020 addresses this topic, but has never been resolved.

And that issue seemed to confuse the (invariable) number of bits in the code, versus the number of digits in its representation, which depends on the base (octal vs hexadecimal).  [I designed cpu's for 20 years, and am pretty familiar with this stuff!]

/Sol.

JohnW4SBA, 2024-03-16:
Sol wrote:
> I will boil down your answers

Thanks Sol. :)

> Acft page should indicate that numeric octal version should be entered in ICAO S code field.
 > Incidentally, Issue #383 from Aug 6, 2020...

I agree - sort of.  First off, clearly my response was fortunately far 
more than you needed - but my hope was to toss enough words at it to 
possibly be helpful for you and anyone else following this topic.  I 
also agree that a small change in the text on that screen, or at least 
in the app Help, would be nice.  But consider two things:
1. Avare is completely free, created and maintained by unpaid volunteers 
operating with the motto: "Do what you want, when you want."  This has 
produced both an exceptional and very popular app, and occasional 
frustration for anyone used to commercial fee-based apps with paid 
employees.

2. The current announced plan is for Avare to be replaced by AvareX in 
the near future, and all volunteer efforts are directed at that quite 
ambitious (again un-reimbursed) goal.

> I designed cpu's for 20 years, and am pretty familiar with this stuff!

Thanks for helping to clarify this for anyone unclear about how to input 
the S Code or what it's used for.  Having this topic in the Forum may 
help others in the future, and might also be considered by the Dev Team 
when coding the new AvareX app.

Happy Flying
John

SolKatzman, 2024-03-24:
Update:

I went flying after entering the Octal (Base 8) value into the "ICAO mode S code" field of the Acft page in Avare.
I made sure that my Acft was "loaded".

Unfortunately, my "ownship" still shows up as trafffic from the ADSB-In device (uAvionix SkySensor). I will
close this issue, update issue #383 and open a new issue about the "ownship" traffic display, which I hope
the AvareMP team will be able to address.

SolKatzman, 2024-03-24:
If the requirement to input the Octal version of the ICAO mode S code is documented, there is no need to allow hexadecimal digits to be entered.

closed 2024-03-24 by SolKatzman