apps4av/avare #510: Unable to update database

xxxxme, 2024-07-20:
When attempting to download/update maps, this fails with the message, "Download Failed: databases, code unable to download the file from this cycle".

Avare 10.3.2.328 from F-Droid
LineageOS 20 (A14) on Pixel 5a
Also happens on Samsung SM-T870 w/ OneUI 5.1.1 (A13).
Happens on wifi and cellular data.

apps4av, 2024-07-20:
Version 11 is the latest. Update to it to get charts.

xxxxme, 2024-07-20:
Installed version 11. Now when zooming in, the app demands, "Download Sectional Chicago" (for example), but that's no longer an option in the download menu. "North Central" *is* downloaded.

n251ea, 2024-07-20:
These questions should be on the mailing list not GitHub and has been answered there previously.

The reason you are getting the error is because Chicago is in the East Central charts,  we are using the same division that the FAA uses for the chart supplements.

closed 2024-07-20 by xxxxme